djwillis/meta-raspberrypi #103: Compile kernel fails

zettelmj, 2015-01-14:
Hi,

i'm trying to compile everything under linux (which i know is not supported). Appart from compiling the kernel there is no problem. The kernel always gives me:

NOTE: Executing RunQueue Tasks
ERROR: Function failed: do_configure (log file is located at /home/jens/raspberrypi/yocto/poky/raspberrypi/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/3.12.26+gitf03cd5e1012d3fe8314a944879308cf0f3d9e29b-r0/temp/log.do_configure.18715)
ERROR: Logfile of failure stored in: /home/jens/raspberrypi/yocto/poky/raspberrypi/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/3.12.26+gitf03cd5e1012d3fe8314a944879308cf0f3d9e29b-r0/temp/log.do_configure.18715
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: Executing shell function do_configure
| NOTE: make oldconfig
| make: *** No rule to make target 'oldconfig'.  Stop.
| ERROR: oe_runmake failed
| WARNING: /home/jens/raspberrypi/yocto/poky/raspberrypi/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/3.12.26+gitf03cd5e1012d3fe8314a944879308cf0f3d9e29b-r0/temp/run.do_configure.18715:1 exit 1 from
|   oe_runmake oldconfig
| ERROR: Function failed: do_configure (log file is located at /home/jens/raspberrypi/yocto/poky/raspberrypi/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/3.12.26+gitf03cd5e1012d3fe8314a944879308cf0f3d9e29b-r0/temp/log.do_configure.18715)
ERROR: Task 5 (/home/jens/raspberrypi/yocto/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_3.12.26.bb, do_configure) failed with exit code '1'

any ideas on that? Or am i running into a problem cause i'm ignoring:

WARNING: Host distribution "Gentoo" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.?

Cheers
Jens

zhangping, 2015-01-15:
I encounter the same problem

zettelmj, 2015-01-15:
I found a solution

apparently there was a change in an inc file (commit is 106dab2fd0321e6b4e77b40111e59a3a31d329d4) search for a patch named:
meta-oe-linux.inc-support-building-with-B-S-which-is-default-now.patch

i just adapted those changes to this repo and now the kernel compiles i haven't tried starting it yet though!

Thank you Martin Jansa for the patch file!

once i get around setting up git and stuff correctly i supplie a patch.

Cheers

diff --git a/recipes-kernel/linux/linux-raspberrypi/defconfig b/recipes-kernel/linux/linux-raspberrypi/defconfig
new file mode 100644
index 0000000..2cd8227
--- /dev/null
+++ b/recipes-kernel/linux/linux-raspberrypi/defconfig
@@ -0,0 +1,1123 @@
+# CONFIG_ARM_PATCH_PHYS_VIRT is not set
+CONFIG_PHYS_OFFSET=0
+# CONFIG_LOCALVERSION_AUTO is not set
+CONFIG_SYSVIPC=y
+CONFIG_POSIX_MQUEUE=y
+CONFIG_FHANDLE=y
+CONFIG_AUDIT=y
+CONFIG_NO_HZ=y
+CONFIG_HIGH_RES_TIMERS=y
+CONFIG_BSD_PROCESS_ACCT=y
+CONFIG_BSD_PROCESS_ACCT_V3=y
+CONFIG_TASKSTATS=y
+CONFIG_TASK_DELAY_ACCT=y
+CONFIG_TASK_XACCT=y
+CONFIG_TASK_IO_ACCOUNTING=y
+CONFIG_IKCONFIG=y
+CONFIG_IKCONFIG_PROC=y
+CONFIG_CGROUP_FREEZER=y
+CONFIG_CGROUP_DEVICE=y
+CONFIG_CGROUP_CPUACCT=y
+CONFIG_RESOURCE_COUNTERS=y
+CONFIG_MEMCG=y
+CONFIG_BLK_CGROUP=y
+CONFIG_NAMESPACES=y
+CONFIG_SCHED_AUTOGROUP=y
+CONFIG_BLK_DEV_INITRD=y
+CONFIG_EMBEDDED=y
+# CONFIG_COMPAT_BRK is not set
+CONFIG_PROFILING=y
+CONFIG_OPROFILE=m
+CONFIG_KPROBES=y
+CONFIG_JUMP_LABEL=y
+CONFIG_MODULES=y
+CONFIG_MODULE_UNLOAD=y
+CONFIG_MODVERSIONS=y
+CONFIG_MODULE_SRCVERSION_ALL=y
+CONFIG_BLK_DEV_THROTTLING=y
+CONFIG_PARTITION_ADVANCED=y
+CONFIG_MAC_PARTITION=y
+CONFIG_CFQ_GROUP_IOSCHED=y
+CONFIG_ARCH_BCM2708=y
+CONFIG_BCM2708_DT=y
+CONFIG_PREEMPT=y
+CONFIG_AEABI=y
+CONFIG_CLEANCACHE=y
+CONFIG_FRONTSWAP=y
+CONFIG_CMA=y
+CONFIG_UACCESS_WITH_MEMCPY=y
+CONFIG_SECCOMP=y
+CONFIG_ZBOOT_ROM_TEXT=0x0
+CONFIG_ZBOOT_ROM_BSS=0x0
[... 1120 lines cut ...]
     else
-        echo "CONFIG_$1=$2" >> ${S}/.config
+        echo "CONFIG_$1=$2" >> ${B}/.config
     fi
 }
 
 do_configure_prepend() {
     # Clean .config
-    echo "" > ${S}/.config
+    echo "" > ${B}/.config
     CONF_SED_SCRIPT=""
 
     # oabi / eabi support
@@ -108,9 +108,9 @@ do_configure_prepend() {
 
     # Keep this the last line
     # Remove all modified configs and add the rest to .config
-    sed -e "${CONF_SED_SCRIPT}" < '${WORKDIR}/defconfig' >> '${S}/.config'
+    sed -e "${CONF_SED_SCRIPT}" < '${WORKDIR}/defconfig' >> '${B}/.config'
 
-    yes '' | oe_runmake oldconfig
+    yes '' | oe_runmake -C ${S} O=${B} oldconfig
 }
 
 # Automatically depend on lzop-native if CONFIG_KERNEL_LZO is enabled

agherzan, 2015-01-18:
Thanks. Well, strangely I didn't run into these errors. But it makes sense and I'll try to tonight to update stuff and shoot a build. Please use the mailing list for other discussions. Thanks in advance.

agherzan, 2015-01-25:
Fixed in master.

closed 2015-01-25 by agherzan