djwillis/meta-raspberrypi #104: Enable Hard Float Support ?

andreabenini, 2015-01-17:
Hi guys,
I'd like to know if your layer could support armhf, if it supports it I'd like to know how to enable it, I'm currently generating armel binaries

toolmmy, 2015-01-17:
As far as is can see it in the conf/machine/include/tune-arm1176jzf-s.inc file you can set your default tune to "arm1176jzfshf"
You can do this in your local.conf: 

conf/local.conf:
DEFAULTTUNE = "arm1176jzfshf"

our you can add a second machine configuration (e.g. raspberrypi-hf.conf).
Haven't tried it for my own so would be great if you could keep us in touch.

agherzan, 2015-01-18:
Indeed. There is already a mailing list thread opened on this topic. Let's not have different threads for the same issue. As README states, we are using mailing lists for any discussions, so, let's follow up there.

Here is the ml thread link: https://lists.yoctoproject.org/pipermail/yocto/2015-January/022983.html

Thanks a lot for you support.

closed 2015-01-18 by agherzan